linorobot/linorobot2_hardware #3: NativeEhernet failed

aeksiri, 2021-12-27:
*** [.pio/build/hex/libf44/NativeEthernet/NativeEthernet.cpp.o] Error 1
============================================================================ [FAILED] Took 2.63 seconds ============================================================================

Environment    Status    Duration
-------------  --------  ------------
teensy41       SUCCESS   00:00:02.563
teensy40       FAILED    00:00:03.021
teensy36       FAILED    00:00:02.642
teensy35       FAILED    00:00:02.640
teensy31       FAILED    00:00:02.632
dev            FAILED    00:00:02.094
beebo          FAILED    00:00:02.672
beebo_m        FAILED    00:00:02.627
hex            FAILED    00:00:02.629

grassjelly, 2022-01-02:
can you describe what the error is?

Sylvania2, 2022-01-13:
Same error when running: pio run --target upload -e teensy40 ?

`Compiling .pio/build/teensy36/libab6/NativeEthernet/NativeDns.cpp.o
Compiling .pio/build/teensy36/libab6/NativeEthernet/NativeEthernet.cpp.o
Compiling .pio/build/teensy36/libab6/NativeEthernet/NativeEthernetClient.cpp.o
Compiling .pio/build/teensy36/libab6/NativeEthernet/NativeEthernetServer.cpp.o
/home/ubuntu/.platformio/packages/framework-arduinoteensy/libraries/NativeEthernet/src/NativeEthernet.cpp: In static member function 'static int EthernetClass::begin(uint8_t*, long unsigned int, long unsigned int)':
/home/ubuntu/.platformio/packages/framework-arduinoteensy/libraries/NativeEthernet/src/NativeEthernet.cpp:126:30: error: 'FNET_CPU_ETH0_IF' was not declared in this scope
           if(fnet_netif_init(FNET_CPU_ETH0_IF, mac, 6) != FNET_ERR){
                              ^
/home/ubuntu/.platformio/packages/framework-arduinoteensy/libraries/NativeEthernet/src/NativeEthernet.cpp: In static member function 'static void EthernetClass::begin(uint8_t*, IPAddress, IPAddress, IPAddress, IPAddress)':
/home/ubuntu/.platformio/packages/framework-arduinoteensy/libraries/NativeEthernet/src/NativeEthernet.cpp:258:30: error: 'FNET_CPU_ETH0_IF' was not declared in this scope
           if(fnet_netif_init(FNET_CPU_ETH0_IF, mac, 6) != FNET_ERR){
                              ^
*** [.pio/build/teensy36/libab6/NativeEthernet/NativeEthernet.cpp.o] Error 1
============================================== [FAILED] Took 53.11 seconds ==============================================

Environment    Status    Duration
-------------  --------  ------------
teensy36       FAILED    00:00:53.114
========================================= 1 failed, 0 succeeded in 00:00:53.114 =========================================
`

Sylvania2, 2022-01-15:
pio run --target upload -e teensy41 is Compiling without error.
[https://cpperror.com/error/1849055](url)
Deleting package "NativeEthernet" from .platformio/packages/framework-arduinoteensy.. Now pio run --target upload -e teensy40 compiles witout error.
Is there another/better solution ?

best regards

grassjelly, 2022-01-16:
Thanks for reporting the bug. Fixed by this: https://github.com/linorobot/linorobot2_hardware/commit/09a905787ab645cedbfea9c404ed95d281877700

Used lib_ignore to ignore 'NativeEthernet' library except teensy41

grassjelly, 2022-01-16:
@Sylvania2 please close if this resolves your is issue. Thanks

closed 2022-01-29 by grassjelly