linorobot/linorobot2_hardware #4: The firmware was uploaded but nothing's happening.

Shromm-Gaind, 2022-01-25:
I have followed the guide on how to configure the hardware of linorobot, and got past the point of uploading firmware onto the Teensy 4.0 (everything up to this point is successful). However, once I try and drive it around and enter the keys to move it forward and backward nothing happens. I have looked at the troubleshooting guide, but I am very confident I am using the Teensy 4.0.

grassjelly, 2022-01-29:
Could you describe what's your setup like (OS, ROS2 Distro)? Are you running any of the OS on a VM?

Shromm-Gaind, 2022-01-29:
I am using Ubuntu 20.04.1 and ROS2 Foxy. I am able to get to the point on the terminal where it shows me the keys and how to operate it. The motors are spinning when I am using the calibration software and in the right orientation.

grassjelly, 2022-01-29:
Is the Teensy’s LED flashing even without pressing any keys?

Shromm-Gaind, 2022-01-29:
It isn't flashing.

Shromm-Gaind, 2022-01-29:
The LED is on when this command is run ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0.

grassjelly, 2022-03-16:
Could you try uploading a simple microros 'hello world' firmware using this https://github.com/linorobot/micro_ros_platformio. Use the same platformio config as linorobot2. 

It's supposed to publish an incrementing number on `micro_ros_arduino_node_publisher` topic.

Cheason28, 2022-03-24:
Thank you for your advice! We've managed to upload the "hello world" firmware. Unfortunately, the nodes are not being set up and the topics are not advertised as shown in these screenshots. We appreciate any advice on how to proceed
[image]
[image]
[image]
.

grassjelly, 2022-03-25:
Looks like you have one more device in the /dev/ttyACM* port. Can you confirm that you are assigning the correct serial port?

Shromm-Gaind, 2022-04-13:
I am still having the issue. Where the calibration when run using screen /dev/ttyACM0. I get the wheels to spin forward, I get the encoders values etc. However when I try and run telop nothing is happening, it doesn't even move forward.  Is it possible I am configuring it wrong?

grassjelly, 2022-04-13:
I've updated the firmware recently, could you pull and try the latest version?

Shromm-Gaind, 2022-04-13:
Will do so, and let you know. Is it possible that the reason for the robot to not move after the calibration is likely due to micro_ros not communicating with the teensy board. Because as far as I can see the calibration doesn't require micro_ros.

grassjelly, 2022-04-13:
yes could be related to this https://github.com/linorobot/linorobot2_hardware/commit/5220465a93f59b0763719397de1c76aa076b1e59

grassjelly, 2022-04-17:
hey @Shromm-Gaind. Did my previous answer resolve your issue? Could you try uploading the most recent firmware

Shromm-Gaind, 2022-04-17:
I have done so, but teleop_twist_keyboard still doesn't work. I think it is just micro_ros, but when I run the agent, there are no bugs or problems.

grassjelly, 2022-04-17:
are you running teleop_twist_keyboard from the same machine where the agent resides?

Shromm-Gaind, 2022-04-17:
Yes

grassjelly, 2022-04-17:
does the Teensy's LED flash (fast rate) when you press a control key from teleop_twist_keyboard

Shromm-Gaind, 2022-04-17:
No, should it?

grassjelly, 2022-04-17:
Just curious aside from the robot's Teensy controller, are there any USB device connected to the robot's computer? If so, could you remove everything except the Teensy controller? Just to ensure there's no serial port clash or anything.

Shromm-Gaind, 2022-04-17:
No, there are no other serial connections

grassjelly, 2022-04-17:
Alright, it seems that the agent is not picking up the Twist msgs. The Teensy's LED is supposed to flash when Twist commands are received. Could you do this simple agent test https://github.com/linorobot/micro_ros_platformio and check if you're able to echo the data.

Shromm-Gaind, 2022-04-17:
I believe that it might be my Teensy having some problem at this point. I have a spare one at my uni. Will try that out and let you know.

grassjelly, 2022-04-17:
Alright, let me know how the agent test go. Thanks

Shromm-Gaind, 2022-04-18:
I have managed to solve the issue, I did replace the teensy, but that did not work as expected. But, I swapped from using a Intel Nuc to a raspberry pi 4 and it seems to work.

Shromm-Gaind, 2022-04-18:
I should have checked earlier if the Nuc was compatible, my fault. Thanks a lot with the help though.

grassjelly, 2022-04-18:
Thanks! Closing this for now

closed 2022-04-18 by grassjelly