linorobot/linorobot2_hardware #6: Odometry topic is empty

crimean7, 2022-03-22:
Hey guys,

I really need your assistance to resolve the issue I have experiencing.

Architecture:
- Jetson Nano
- ROS2 Galactic (same issue with Foxy)
- Teensy 3.2 (same issue with Teensy 3.5)
- IMU: MPU6050
- Brushless Motors with built-in encoders (max RPM 500, rated RPM 400)

The Issue:
    ros2 topic echo /odom/unfiltered  NO DATA PUBLISHED

Thanks in advance for your help

kajMork, 2022-04-01:
I have the same problem.
Setup is the same, except that I use a Teensy 4.1 and this [Pololu motor](https://www.pololu.com/product/4752).
Calibration process for encoder is also working fine for me, and other topics such as "imu/data" and "cmd_vel" works fine.

kajMork, 2022-04-06:
I solved the problem by changing rclc_publisher_init_best_effort to rclc_publisher_init_default at where the odometry publisher is created.
So from:
``` C++
    // create odometry publisher
    RCCHECK(rclc_publisher_init_best_effort( 
        &odom_publisher, 
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(nav_msgs, msg, Odometry),
        "odom/unfiltered"
    ));
``` 
To
``` C++
    // create odometry publisher
    RCCHECK(rclc_publisher_init_default( 
        &odom_publisher, 
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(nav_msgs, msg, Odometry),
        "odom/unfiltered"
    ));
```

closed 2022-04-13 by grassjelly