linorobot/linorobot2_hardware #7: L298n driver to move forward and backward with teleop

manirajanvn, 2022-04-11:
Hello,
I m using teensy 4.0, L298N(GENERIC_2_IN_MOTOR_DRIVER ), GY-85 and rplidar. After upload firmware to teensy, was using teleop to test. when I press **_u, i, o_** wheels are rotating in forward and when I press **_m, ,(comma), .(dot)_** wheels are not rotating in backward. Just want to know is this expected or Am I missing something. Please find the attached driver image and config

Thanks
Mani

[image]
i
[image]
[lino_base_config.h.txt](https://github.com/linorobot/linorobot2_hardware/files/8461144/lino_base_config.h.txt)

grassjelly, 2022-04-11:
Hey @manirajanvn , 
 
Are the wheels not spinning backwards during calibration too or this behavior only happens with the firmware (w/ microros)?

manirajanvn, 2022-04-11:
Hi,
Wheels are rotating forward separately for spin and sample commands during calibration.

manirajanvn, 2022-04-16:
Hi,
This issue is resolved, have missed L298's GND to Teensy.
Thanks

closed 2022-04-16 by manirajanvn