linorobot/linorobot2_hardware #9: Autoconnect Failed Between Teensy 4.1 and ROS2 Foxy

mich1342, 2022-05-25:
I am using Intel NUC with Ubuntu 20.04 ROS2 Foxy connected with Teensy 4.1 running code from this repo. 

I have modified the code so it could run in Arduino IDE and adjust the components code (my IMU and BLDC driver is not supported yet by linorobot2). I got  re-connection issues when running the agent. Most of the time I need to re-upload the code to the Teensy when the agent is running. 

Comparing the loop in this repo and the loop inside micro-ROS re-connection example, I added delay (around 100-500 ms) at the end of the loop and solved the re-connection issues. At the end I could go as low as 10-20 ms delay. Is there any other solution for this problem since I do not think adding delay is a good solution in terms of the robot responses.

[image]

the image above shows the agent terminal output when running the agent (The Teensy is always connected to the NUC). the topic that should be published by the Teensy sometimes appears when I run the "ros2 topic list" command but nothing published.

[image]

the image above shows the agent terminal output after re-uploading the agent. After this shows up, everything run as it should be.

Thank you

grassjelly, 2022-05-28:
Hi, are you able to get the same results when using fake imu? (See more [here](https://github.com/linorobot/linorobot2_hardware/blob/master/config/custom/dev_config.h#L23)) Just to make sure, it’s not related to hardware

mich1342, 2022-05-28:
Hi, I have not tried that yet, but I have tried make the code consist of the cmd_vel subscribers only. It does not publishing any IMU or odom data. I do not even create the executor and publisher of them. I got exactly the same result. 

I thought the cause of the problem is your line of code here 

https://github.com/linorobot/linorobot2_hardware/blob/e80af875113f80bcd52b608e194d96836ac3d9c8/firmware/src/firmware.ino#L132

Due to the Teensy 4.1 high clock, I though the spin some node is not finish executed yet or has not reach the 10ms timeout but already called and cause some bugs?

I have checked the newest micro-ROS repo for the auto reconnect example. They are not using the same loop frame as you are using in your current code.

As a supporting argument of  my suspicion, I modify you loop structure so that the spin some function is called at least 10ms after the last call done. It solved the problem. The old micro-ROS example also use some small delays to make sure those function are called at least after several seconds after previous call. 

https://github.com/micro-ROS/micro_ros_arduino/blob/4998c25f0fb502380ac6a7438daba1ff9c0c9771/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L108

aeksiri, 2022-05-31:
> Hi, are you able to get the same results when using fake imu? (See more [here](https://github.com/linorobot/linorobot2_hardware/blob/master/config/custom/dev_config.h#L23)) Just to make sure, it’s not related to hardware

I have been tried to use the FAKE IMU, the result still the same (the "ros2 topic list" command but nothing published.).

mich1342, 2022-05-31:
> > Hi, I have not tried that yet, but I have tried make the code consist of the cmd_vel subscribers only. It does not publishing any IMU or odom data. I do not even create the executor and publisher of them. I got exactly the same result.
> 
> > 
> 
> > I thought the cause of the problem is your line of code here
> 
> > 
> 
> > https://github.com/linorobot/linorobot2_hardware/blob/e80af875113f80bcd52b608e194d96836ac3d9c8/firmware/src/firmware.ino#L132
> 
> > 
> 
> > Due to the Teensy 4.1 high clock, I though the spin some node is not finish executed yet or has not reach the 10ms timeout but already called and cause some bugs?
> 
> > 
> 
> > I have checked the newest micro-ROS repo for the auto reconnect example. They are not using the same loop frame as you are using in your current code.
> 
> > 
> 
> > As a supporting argument of my suspicion, I modify you loop structure so that the spin some function is called at least 10ms after the last call done. It solved the problem. The old micro-ROS example also use some small delays to make sure those function are called at least after several seconds after previous call.
> 
> > 
> 
> > https://github.com/micro-ROS/micro_ros_arduino/blob/4998c25f0fb502380ac6a7438daba1ff9c0c9771/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L108
> 
> 
> 
> 
> 
> I also have been tried to add the "rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));", unfortunately, I have no luck.

Hi, try to run those command every 10-20ms using the millis() function

mich1342, 2022-05-31:
> Hi michael
> please make sure the IMU is working properly.
> and make sure that the magnetometer working.
> in my case, because the case from metal, the localization seems bit error
> while running navigation.
> and sometimes need several restart to make the IMU working.
> 
> and one thing, idk is it necessary there in linorobot 2, in linorobot 1 you
> need to calibrate the imu first before using it.
> 
> hope helpfull
> 
> On Tue, May 31, 2022 at 6:30 PM Akkharaphong EKSIRI <
> 
> > Hi, I have not tried that yet, but I have tried make the code consist of
> > the cmd_vel subscribers only. It does not publishing any IMU or odom data.
> > I do not even create the executor and publisher of them. I got exactly the
> > same result.
> >
> > I thought the cause of the problem is your line of code here
> >
> >
> > https://github.com/linorobot/linorobot2_hardware/blob/e80af875113f80bcd52b608e194d96836ac3d9c8/firmware/src/firmware.ino#L132
> >
> > Due to the Teensy 4.1 high clock, I though the spin some node is not
> > finish executed yet or has not reach the 10ms timeout but already called
> > and cause some bugs?
> >
> > I have checked the newest micro-ROS repo for the auto reconnect example.
> > They are not using the same loop frame as you are using in your current
> > code.
> >
> > As a supporting argument of my suspicion, I modify you loop structure so
> > that the spin some function is called at least 10ms after the last call
> > done. It solved the problem. The old micro-ROS example also use some small
> > delays to make sure those function are called at least after several
> > seconds after previous call.
> >
> >
> > https://github.com/micro-ROS/micro_ros_arduino/blob/4998c25f0fb502380ac6a7438daba1ff9c0c9771/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L108
> >
> > I also have been tried to add the "rclc_executor_spin_some(&executor,
> > RCL_MS_TO_NS(10));", unfortunately, I have no luck.

Hi, I have tried and make sure that my IMU is working. I am using BNO080 IMU, the calibration process already done outside ROS. I think it only use mainly gyroscope to calculate the changes in headings ?

grassjelly, 2022-06-04:
Hey @mich1342 . I can't seem to replicate your reconnection problem. Btw the firmware may take a few seconds to reconnect as mentioned here https://github.com/micro-ROS/micro_ros_arduino/issues/659#issuecomment-1000086435 . Could it be you're reuploading the firmware too soon? Meanwhile I'll try to continue replicating the error.

grassjelly, 2022-06-04:
hey @aeksiri . Can you confirm you have the latest version of the `following:`
- Latest ROS packages (`sudo apt update && sudo apt upgrade`)
- Latest microrosarduino library (on linorobot2_hardware/firmware) run: `pio lib update `
- Latest microROS agent: `cd <your_ws>/src/uros/micro-ROS-Agent && git pull` and follow these [instructions](https://github.com/linorobot/linorobot2/blob/foxy/ROBOT_INSTALLATION.md#15-setup-micro-ros-agent). Don't forget to builld and source your workspace.

grassjelly, 2022-06-22:
hey @mich1342. Updated the main loop to sync with microROS reconnection examples. Does that work for your too?

mich1342, 2022-07-05:
Hi @grassjelly sorry for the late update. It works perfectly. One of my biggest problem is my Inverter. Everything works perfectly after removing the inverter. Thank you

closed 2022-07-05 by mich1342