linorobot/linorobot2_hardware #12: Support for Teensy 3.2?

AsherThomasBabu, 2022-05-29:
The readme for the hardware part does'nt mention support for Teensy 32, but the original linobot does.
I was looking to port my old hardware to linobot2. Was the 3.2 missed out? Or is it no longer supported?

grassjelly, 2022-05-29:
Hey @AsherThomasBabu . It is still supported just use teensy31 as en environment

AsherThomasBabu, 2022-05-30:
Thank you @grassjelly . I tried this on multiple operating systems, I tried out both the images of 16.04 and 20.04 from Ubiquity Robots. But could'nt get my teensy to show up on /dev/ttyACM0. 

I did program the teensy with the fast_blink using my mac before connecting it to the Pi. 
I also did copy the udev rules from the pjrc website to the specified location

Is this something that you have come across, have been stuck on this for the past week, would be grateful for any inputs

AsherThomasBabu, 2022-06-07:
@grassjelly Yes, I did. in fact I tried using a different teensy which showed up as linobase once udev was setup, but disappeared after I uploaded the firmware.

grassjelly, 2022-06-22:
This usually happens when the Teensy is new. Most importantly, your machine should be able to detect the board when you run `lsusb` then you can upload with `pio run --target upload -e teensy31 `. The device should then show as /dev/ttyACM*