linorobot/linorobot2_hardware #13: micro_ros_utilities/type_utilities.h: No such file or directory

hwacoding, 2022-06-30:
Getting the following error when I try to load the firmware to a teensy41.

Compiling .pio/build/teensy41/lib443/imu/ADXL345.cpp.o
In file included from /home/jetson/linorobot2_hardware/firmware/src/firmware.ino:32:0:
lib/odometry/odometry.h:19:48: fatal error: micro_ros_utilities/type_utilities.h: No such file or directory
compilation terminated.
Compiling .pio/build/teensy41/lib443/imu/HMC5883L.cpp.o
*** [.pio/build/teensy41/src/firmware.ino.cpp.o] Error 1
========================= [FAILED] Took 17.60 seconds =========================

Environment Status Duration

teensy41 FAILED 00:00:17.599
==================== 1 failed, 0 succeeded in 00:00:17.599 ====================

Please advise. Thank you

from linorobot2.

grassjelly, 2022-07-01:
hey @hwacoding do you mind sharing what caused the problem?

hwacoding, 2022-07-01:
Already solved it. Thanks

grassjelly, 2022-07-01:
hey @hwacoding . Could you share how it was solved please in case other users face the same problem.

adityasivaraj, 2022-07-02:
Hey @hwacoding , I am also facing the same issue, could you let me know how you solved it?

hwacoding, 2022-07-04:
oh sorry, I am late.
firmware/platformio.ini
I add lib_deps =  https://github.com/micro-ROS/micro_ros_utilities

And, we need rosidl.
https://github.com/ros2/rosidl
I install these in my computer.
And paste it firmware/.pio/libdeps/teensy40

adityasivaraj, 2022-07-04:
hey @hwacoding , Thank you for getting back. Both of these changes in robot computer?

adityasivaraj, 2022-07-04:
Also, for some reason rosidl is not recognized at all by platformio... could you tell me how do you install rosidl?

grassjelly, 2022-07-04:
I can't replicate this problem. @adityasivaraj could you totally remove your linorobot2_hardware, clone and reinstall the firmware?

adityasivaraj, 2022-07-04:
Hey @grassjelly , so I think I solved the problem. I added micro_ros_utilities as mentioned in [here](https://github.com/linorobot/linorobot2_hardware/issues/13#issuecomment-1173432705). Then I cloned the foxy build of rosidl into linorobot2_ws/src, then rosdep install --from-paths src -y --ignore-src. colcon build after that, and copy rosidl/rosidl_typesupport_introspect_c folder to firmware/.pio/libdeps/teensy31/. This seemingly works for pio upload. But now teleop commands are not working, and I am not sure if the two are related.

Did this problem occur because I cloned linorobot2_hardware in linorobot2_ws/src? The instructions were not clear on where to install this repo.

adityasivaraj, 2022-07-04:
> Could you try if setting this line https://github.com/linorobot/linorobot2_hardware/blob/master/firmware/platformio.ini#L16 to galactic fixes all your problem?

No, unfortunately that did not solve my issue.

> Meanwhile, have you considered moving to galactic?

Not right now, I want to develop with something that has long term support, I tried running galactic build in humble installation, but that had a lot of dependency issues...

grassjelly, 2022-07-04:
hey @adityasivaraj could you try the recent commits on foxy branch. Added new installation guide here: https://github.com/linorobot/linorobot2_hardware#2-download-linorobot2_hardware

adityasivaraj, 2022-07-05:
Hey @grassjelly , Yes this works! Although there maybe a problem with the calibration firmware upload, I am not sure if I screwed up while installing, but, when I copied the .pio folder from the main firmware folder to calibration folder, it worked fine.

closed 2022-07-05 by grassjelly