linorobot/linorobot2_hardware #15: micro_ros_agent stuck at logger setup

Hernansidris, 2022-07-14:
Hello

when i run this 
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0

[image]

there is no 
| Root.cpp | create_client | create
| SessionManager.hpp | establish_session | session established
only I can see Root.cpp.

i already reconnect the teensy and the result same

[image]

i also run this 
ros2 topic list

[image]

there is no topic publish 
_example_
/imu/data
/odom/unfiltered
/parameter_events
/rosout

is there another/better solution?

closed 2022-07-14 by Hernansidris

grassjelly, 2022-07-14:
hey @Hernansidris do you mind sharing what caused the problem?

Hernansidris, 2022-07-14:
hay @grassjelly 
the problem i got is imu connection failed

this is not best solution but i just bypass imu connection in sourcode linorobot2_hardware/firmware.ino

[image]

the micro_ros agent not stuck again and imu data work
i dont know what the real problem

shazib2t, 2024-02-08:
I know it is old post but I had the same issue and found the IMU was not correctly selected in the lino_base_config.h