linorobot/linorobot2_hardware #19: ESP32 build error lib/imu/I2Cdev.cpp:278:75: error: no matching function for call to 'min(uint8_t&, int)'

samuk, 2022-07-26:
Appreciate ESP32 is not officially supported.

I get 

`lib/imu/I2Cdev.cpp:278:75: error: no matching function for call to 'min(uint8_t&, int)'`

Which I think is related to https://github.com/RobTillaart/Arduino/issues/128

Line 278 of lib/imu/I2Cdev.cpp is

`      for (uint8_t k = 0; k < length; k min(length, BUFFER_LENGTH)) {`

If I add an underscore to min that error goes away but I get 

`/home/ubuntu/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:206:35: error: 'k' cannot be used as a function`

pablogs9, 2022-08-04:
`clock_gettime` is being defined [here](https://github.com/micro-ROS/micro_ros_platformio/blob/b256be95073d01fd06ba5f97dbb7c312e3dc697c/platform_code/arduino/clock_gettime.cpp#L6) since it is a POSIX function that is not normally available in the embedded targets.

If your platform is providing it you should remove this micro-ROS implementation. If you solve it using some kind of conditional compilation based on `#ifdef` please contribute with a PR to the micro-ROS for PlatformIO repo.

pablogs9, 2022-08-04:
In any case, I have tried to replicate:

```bash
git clone -b galactic https://github.com/linorobot/linorobot2_hardware
cd linorobot2_hardware/firmware
pio run --target upload -e ESP32
```
and I get some compilation errors on this proyect:

```
In file included from lib/imu/MPU9250.cpp:37:
lib/imu/MPU9250.h:60:24: error: conflicting declaration 'typedef int8_t prog_int8_t'
         typedef int8_t prog_int8_t;
                        ^~~~~~~~~~~
In file included from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:29,
                 from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:165,
                 from lib/imu/I2Cdev.h:91,
                 from lib/imu/MPU9250.h:40,
                 from lib/imu/MPU9250.cpp:37:
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:25:14: note: previous declaration as 'typedef char prog_int8_t'
 typedef char prog_int8_t;
              ^~~~~~~~~~~
In file included from lib/imu/MPU9250.cpp:37:
lib/imu/MPU9250.h:64:25: error: conflicting declaration 'typedef int32_t prog_int32_t'
         typedef int32_t prog_int32_t;
                         ^~~~~~~~~~~~
In file included from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:29,
                 from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:165,
                 from lib/imu/I2Cdev.h:91,
                 from lib/imu/MPU9250.h:40,
                 from lib/imu/MPU9250.cpp:37:
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:29:14: note: previous declaration as 'typedef long int prog_int32_t'
 typedef long prog_int32_t;
              ^~~~~~~~~~~~
In file included from lib/imu/MPU9250.cpp:37:
lib/imu/MPU9250.h:65:26: error: conflicting declaration 'typedef uint32_t prog_uint32_t'
         typedef uint32_t prog_uint32_t;
                          ^~~~~~~~~~~~~
In file included from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:29,
                 from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:165,
                 from lib/imu/I2Cdev.h:91,
                 from lib/imu/MPU9250.h:40,
                 from lib/imu/MPU9250.cpp:37:
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:30:23: note: previous declaration as 'typedef long unsigned int prog_uint32_t'
 typedef unsigned long prog_uint32_t;
```

I guess that there are some configuration steps previous to the build step. So, could you provide some replication steps?

pablogs9, 2022-08-04:
Check this branch and let me know if this solves the issue: https://github.com/micro-ROS/micro_ros_platformio/pull/48

samuk, 2022-08-04:
Great thanks that worked.

Linking .pio/build/default/firmware.elf
Retrieving maximum program size .pio/build/default/firmware.elf
Checking size .pio/build/default/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.4% (used 43772 bytes from 327680 bytes)
Flash: [===       ]  29.3% (used 383489 bytes from 1310720 bytes)
Building .pio/build/default/firmware.bin
esptool.py v3.3
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.

You'd need to use my fork to get it to compile:
git clone -b galactic https://github.com/rosmo-robot/linorobot2_hardware

closed 2022-08-04 by samuk