linorobot/linorobot2_hardware #24: Teensy 4.0 compile errors

EnderDragonEP, 2022-09-28:
I keeps getting this errors while compile the firmware for teensy 4.0.
And topics `/odom/unfiltered` and `/imu/data` are missing after running the micro-ROS agent.

Terminal logs:
<pre>Building micro-ROS library
Found 99 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.22cf9b6
|-- encoder
|-- imu
|   |-- Wire @ 1.0
|   |-- micro_ros_platformio @ 0.0.1+sha.22cf9b6
|-- kinematics
|-- motor
|   |-- Servo @ 1.1.2
|-- odometry
|   |-- micro_ros_platformio @ 0.0.1+sha.22cf9b6
|-- pid
Building in release mode
Compiling .pio/build/teensy40/src/firmware.ino.cpp.o
Compiling .pio/build/teensy40/lib640/micro_ros_platformio/platform_code/arduino/clock_gettime.cpp.o
Compiling .pio/build/teensy40/lib640/micro_ros_platformio/platform_code/arduino/serial/micro_ros_transport.cpp.o
Compiling .pio/build/teensy40/lib81a/encoder/encoder.cpp.o
Compiling .pio/build/teensy40/lib0e1/Wire/Wire.cpp.o
Archiving .pio/build/teensy40/lib640/libmicro_ros_platformio.a
Archiving .pio/build/teensy40/lib81a/libencoder.a
Indexing .pio/build/teensy40/lib640/libmicro_ros_platformio.a
<font color="#C4A000">/home/jetson/linorobot2_hardware/firmware/src/firmware.ino: In function &apos;bool destroyEntities()&apos;:</font>
<font color="#C4A000">/home/jetson/linorobot2_hardware/firmware/src/firmware.ino:222:47: warning: ignoring return value of &apos;rcl_ret_t rcl_publisher_fini(rcl_publisher_t*, rcl_node_t*)&apos;, declared with attribute warn_unused_result [-Wunused-result]</font>
<font color="#C4A000">     rcl_publisher_fini(&amp;odom_publisher, &amp;node);</font>
<font color="#C4A000">                                               ^</font>
<font color="#C4A000">/home/jetson/linorobot2_hardware/firmware/src/firmware.ino:223:46: warning: ignoring return value of &apos;rcl_ret_t rcl_publisher_fini(rcl_publisher_t*, rcl_node_t*)&apos;, declared with attribute warn_unused_result [-Wunused-result]</font>
<font color="#C4A000">     rcl_publisher_fini(&amp;imu_publisher, &amp;node);</font>
<font color="#C4A000">                                              ^</font>
<font color="#C4A000">/home/jetson/linorobot2_hardware/firmware/src/firmware.ino:224:52: warning: ignoring return value of &apos;rcl_ret_t rcl_subscription_fini(rcl_subscription_t*, rcl_node_t*)&apos;, declared with attribute warn_unused_result [-Wunused-result]</font>
<font color="#C4A000">     rcl_subscription_fini(&amp;twist_subscriber, &amp;node);</font>
<font color="#C4A000">                                                    ^</font>
<font color="#C4A000">/home/jetson/linorobot2_hardware/firmware/src/firmware.ino:225:25: warning: ignoring return value of &apos;rcl_ret_t rcl_node_fini(rcl_node_t*)&apos;, declared with attribute warn_unused_result [-Wunused-result]</font>
<font color="#C4A000">     rcl_node_fini(&amp;node);</font>
<font color="#C4A000">                         ^</font>
<font color="#C4A000">/home/jetson/linorobot2_hardware/firmware/src/firmware.ino:226:35: warning: ignoring return value of &apos;rcl_ret_t rcl_timer_fini(rcl_timer_t*)&apos;, declared with attribute warn_unused_result [-Wunused-result]</font>
<font color="#C4A000">     rcl_timer_fini(&amp;control_timer);</font>
<font color="#C4A000">                                   ^</font>
Indexing .pio/build/teensy40/lib81a/libencoder.a
Compiling .pio/build/teensy40/lib0e1/Wire/WireIMXRT.cpp.o
Compiling .pio/build/teensy40/lib0e1/Wire/WireKinetis.cpp.o
Compiling .pio/build/teensy40/lib0e1/Wire/utility/twi.c.o
Compiling .pio/build/teensy40/lib443/imu/ADXL345.cpp.o
Compiling .pio/build/teensy40/lib443/imu/HMC5883L.cpp.o
Archiving .pio/build/teensy40/lib0e1/libWire.a
Indexing .pio/build/teensy40/lib0e1/libWire.a
Compiling .pio/build/teensy40/lib443/imu/I2Cdev.cpp.o
Compiling .pio/build/teensy40/lib443/imu/ITG3200.cpp.o
Compiling .pio/build/teensy40/lib443/imu/MPU6050.cpp.o
Compiling .pio/build/teensy40/lib443/imu/MPU9150.cpp.o
Compiling .pio/build/teensy40/lib443/imu/MPU9250.cpp.o
Compiling .pio/build/teensy40/libe32/kinematics/kinematics.cpp.o
Archiving .pio/build/teensy40/libe32/libkinematics.a
Indexing .pio/build/teensy40/libe32/libkinematics.a
Compiling .pio/build/teensy40/lib813/Servo/Servo.cpp.o
Archiving .pio/build/teensy40/lib813/libServo.a
Indexing .pio/build/teensy40/lib813/libServo.a
Compiling .pio/build/teensy40/libf69/odometry/odometry.cpp.o
<font color="#C4A000">lib/imu/MPU9150.cpp: In member function &apos;bool MPU9150::writeMemoryBlock(const uint8_t*, uint16_t, uint8_t, uint8_t, bool, bool)&apos;:</font>
<font color="#C4A000">lib/imu/MPU9150.cpp:2992:14: warning: &apos;verifyBuffer&apos; may be used uninitialized in this function [-Wmaybe-uninitialized]</font>
<font color="#C4A000">     uint8_t *verifyBuffer;</font>
<font color="#C4A000">              ^</font>
Compiling .pio/build/teensy40/lib7aa/pid/pid.cpp.o
<font color="#C4A000">lib/imu/MPU9250.cpp: In member function &apos;bool MPU9250::writeMemoryBlock(const uint8_t*, uint16_t, uint8_t, uint8_t, bool, bool)&apos;:</font>
[... 100 lines cut ...]
Linking .pio/build/teensy40/firmware.elf
Checking size .pio/build/teensy40/firmware.elf
Building .pio/build/teensy40/firmware.hex
Configuring upload protocol...
AVAILABLE: jlink, teensy-cli, teensy-gui
CURRENT: upload_protocol = teensy-cli
Rebooting...
Uploading .pio/build/teensy40/firmware.hex
Teensy Loader, Command Line, Version 2.2
Read &quot;.pio/build/teensy40/firmware.hex&quot;: 134144 bytes, 6.6% usage
Unable to soft reboot with USB error: Success
Waiting for Teensy device...
 (hint: press the reset button)
Found HalfKay Bootloader
Read &quot;.pio/build/teensy40/firmware.hex&quot;: 134144 bytes, 6.6% usage
Programming................................................................................................................................
Booting
======================== [<font color="#4E9A06"><b>SUCCESS</b></font>] Took 589.07 seconds ========================

<b>Environment</b>    <b>Status</b>    <b>Duration</b>
-------------  --------  ------------
<font color="#06989A">teensy40</font>       <font color="#4E9A06">SUCCESS</font>   00:09:49.073
<font color="#4E9A06">========================= 1 succeeded in 00:09:49.073 =========================</font>
<font color="#4E9A06"><b>jetson@nano</b></font>:<font color="#3465A4"><b>~/linorobot2_hardware/firmware</b></font>$ 
</pre>

grassjelly, 2022-09-28:
is the LED blinking?

EnderDragonEP, 2022-09-28:
> is the LED blinking?

No, the LED stay on. But if I use `teleop_twist_keyboard` to move the robot around, LED start blinking as the serial receive and sending the packages.

EnderDragonEP, 2022-09-28:
Hmm... After I changed to an different IMU but the same model (MPU-9250/6500), the LED start to blink nonstop. What's does the blinking LED means?

bpepino, 2023-04-05:
I am having the same problem.  The light is blinking three times every 2 seconds or so. But when i run  

ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0

ros2 run teleop_twist_keyboard teleop_twist_keyboard "

I cant drive the robot since no topic have been created

bpepino, 2023-04-05:
Any news on this? I got the same errors and  no topics after running the micro-ros agent. Cannot move the robot even though calibration works perfectly . I am using humble.

hippo5329, 2023-05-21:
It could be the IMU failed to initialize. I synced the i2cdev and imu code from upstream i2cdevlib in my port to ESP32. The imu compatibility might be improved. You may try.

EnderDragonEP, 2023-05-21:
> It could be the IMU failed to initialize. I synced the i2cdev and imu code from upstream i2cdevlib in my port to ESP32. The imu compatibility might be improved. You may try.

Thanks for your reply! This fix my issue!