linorobot/linorobot2_hardware #28: problem  while using  humble distro with linorobot2_hardware(galactic branch)

hemsinghb, 2023-05-16:
Hi Every one, Any one tried with RoS2 Humble with Linorobot2_hardware  ?

 I tried  with presently avaiable linorobot2_hardware (galactic branch -presently available)  and failed to make map. When moving robot front and back ots not moving on rviz, however turning left and right is ok.  So the map could not be made . May be issues with ekf or slam config.yaml file with ros2/Ubuntu22.04 etc. If any one found solution please guide. Thank you, ...Hems.

atticusrussell, 2023-08-26:
@hemsinghb I have tested ROS2 Humble with linorobot2_hardware. Can you try the [humble](https://github.com/linorobot/linorobot2_hardware/tree/humble) branch and report back? 

If it works and the issue is resolved, feel free to mark the issue as closed.

hemsinghb, 2023-08-29:
Thank you so much, its working fine.

atticusrussell, 2023-08-29:
@hemsinghb Awesome. Can you click the purple button at the bottom of this issue to close it?

Could you also close the other issues you opened about this topic: #27 and #30 ?