linorobot/linorobot2_hardware #32: [bug] error linking firmware: undefined references

atticusrussell, 2023-06-30:
Working on RPi4 with teensy 4.1 on branch foxy, ROS Humble Ubuntu 22.04. 
Calibration folder works fine, but uploading firmware following README yields errors:
```
atticus@rospi-ssh-session  <catbot>
~/github/linorobot2_hardware/firmware
18:25:32 $ pio run --target upload -e teensy41
Processing teensy41 (board: teensy41; platform: teensy; framework: arduino)
---------------------------------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/micro-ROS/micro_ros_arduino
git version 2.34.1
Cloning into '/home/atticus/.platformio/.cache/tmp/pkg-installing-_g7s561e'...
remote: Enumerating objects: 2162, done.
remote: Counting objects: 100% (2162/2162), done.
remote: Compressing objects: 100% (673/673), done.
remote: Total 2162 (delta 1559), reused 1786 (delta 1451), pack-reused 0
Receiving objects: 100% (2162/2162), 17.20 MiB | 5.32 MiB/s, done.
Resolving deltas: 100% (1559/1559), done.
Updating files: 100% (1963/1963), done.
Library Manager: micro_ros_arduino@2.0.7-iron+sha.c14c4e3 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.18.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.158.0 (1.58) 
 - tool-teensy @ 1.158.0 (1.58) 
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
Converting firmware.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 99 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_arduino @ 2.0.7-iron+sha.c14c4e3
|-- encoder
|-- imu
|-- kinematics
|-- motor
|-- odometry
|-- pid
Building in release mode
Compiling .pio/build/teensy41/src/firmware.ino.cpp.o
Compiling .pio/build/teensy41/libff1/SPI/SPI.cpp.o
Compiling .pio/build/teensy41/lib697/Ethernet/Dhcp.cpp.o
Compiling .pio/build/teensy41/lib697/Ethernet/Dns.cpp.o
Compiling .pio/build/teensy41/lib697/Ethernet/Ethernet.cpp.o
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino: In function 'bool destroyEntities()':
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino:221:23: warning: ignoring return value of 'rcl_ret_t rcl_publisher_fini(rcl_publisher_t*, rcl_node_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  221 |     rcl_publisher_fini(&odom_publisher, &node);
      |     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino:222:23: warning: ignoring return value of 'rcl_ret_t rcl_publisher_fini(rcl_publisher_t*, rcl_node_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  222 |     rcl_publisher_fini(&imu_publisher, &node);
      |     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino:223:26: warning: ignoring return value of 'rcl_ret_t rcl_subscription_fini(rcl_subscription_t*, rcl_node_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  223 |     rcl_subscription_fini(&twist_subscriber, &node);
      |     ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino:224:18: warning: ignoring return value of 'rcl_ret_t rcl_node_fini(rcl_node_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  224 |     rcl_node_fini(&node);
      |     ~~~~~~~~~~~~~^~~~~~~
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino:225:19: warning: ignoring return value of 'rcl_ret_t rcl_timer_fini(rcl_timer_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  225 |     rcl_timer_fini(&control_timer);
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
Compiling .pio/build/teensy41/lib697/Ethernet/EthernetClient.cpp.o
Archiving .pio/build/teensy41/libff1/libSPI.a
Indexing .pio/build/teensy41/libff1/libSPI.a
Compiling .pio/build/teensy41/lib697/Ethernet/EthernetServer.cpp.o
Compiling .pio/build/teensy41/lib697/Ethernet/EthernetUdp.cpp.o
Compiling .pio/build/teensy41/lib697/Ethernet/socket.cpp.o
Compiling .pio/build/teensy41/lib697/Ethernet/utility/w5100.cpp.o
[... 339 lines cut ...]
Compiling .pio/build/teensy41/FrameworkArduino/usb_serial2.c.o
Compiling .pio/build/teensy41/FrameworkArduino/usb_serial3.c.o
Compiling .pio/build/teensy41/FrameworkArduino/usb_touch.c.o
Compiling .pio/build/teensy41/FrameworkArduino/yield.cpp.o
Archiving .pio/build/teensy41/libFrameworkArduino.a
Indexing .pio/build/teensy41/libFrameworkArduino.a
Linking .pio/build/teensy41/firmware.elf
/home/atticus/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: .pio/libdeps/teensy41/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'
/home/atticus/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: .pio/libdeps/teensy41/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xbe): undefined reference to `__locale_ctype_ptr'
/home/atticus/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: .pio/libdeps/teensy41/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x46): undefined reference to `__locale_ctype_ptr'
/home/atticus/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0xb6): undefined reference to `__locale_ctype_ptr'
/home/atticus/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x164): undefined reference to `__locale_ctype_ptr'
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy41/firmware.elf] Error 1
====================================== [FAILED] Took 89.49 seconds ======================================

Environment    Status    Duration
-------------  --------  ------------
teensy41       FAILED    00:01:29.486
================================= 1 failed, 0 succeeded in 00:01:29.486 =================================

```

grassjelly, 2023-07-01:
could you try uploading the galactic branch?

atticusrussell, 2023-07-01:
Thank you for your quick response. Yes, the galactic branch works for me. 

I ended up trying it right after filing the issue when I saw the change for Foxy in the commit history that seemed related to my issue.

grassjelly, 2023-07-01:
great. closing this for now

closed 2023-07-01 by grassjelly