linorobot/linorobot2_hardware #35: firmware error

justinbar, 2023-07-02:
when i compile the firmware teeny40 i keep on getting this error

[image]
[image]

atticusrussell, 2023-07-02:
use the `galactic` branch if you're on a different one currently. I ran into this issue yesterday and using galactic instead fixed it for me and worked fine with my other ros2 humble nodes.

justinbar, 2023-07-02:
so it doesn't work on foxy

atticusrussell, 2023-07-02:
The "galactic" branch of this repository works on ROS humble. I think it will on ROS foxy as well. The branch named "foxy" is broken.

justinbar, 2023-07-02:
how do i use the foxy branch, ros2 foxy is the only foxy i have right now

atticusrussell, 2023-07-02:
I don't believe the "foxy" branch of this repo works right now, but the "galactic" branch of this repo will likely work with the ROS Foxy on your system, regardless of the different names.

justinbar, 2023-07-02:
so i need to install ros 2 galactic to get galactic branch

atticusrussell, 2023-07-02:
no. Just checkout the "galactic" branch of this repository.
```
cd ~
git clone https://github.com/linorobot/linorobot2_hardware.git
cd linorobot2_hardware
git checkout galactic
```

justinbar, 2023-07-02:
why is giving this error 
[image]
[image]

justinbar, 2023-07-03:
when i use checkout galactic and try to run the firmware to my teeny40 i get this error
[image]

justinbar, 2023-07-03:
when i use the galactic branch and try to run the firmware to my teensy40 i get this error
[image]
[image]

atticusrussell, 2023-07-03:
I see. I have a guess on how to fix it, but can't guarantee it will work because I haven't tested myself.

In `firmware/platformio.ini`, try changing
`lib_deps = https://github.com/micro-ROS/micro_ros_platformio`
to 
`lib_deps = https://github.com/micro-ROS/micro_ros_platformio#1.0.0`

Looking at last line of the the traceback, it mentions line 96 of 
[micro_ros_platformio/microros_utils/library_builder.py](https://github.com/micro-ROS/micro_ros_platformio/blob/main/microros_utils/library_builder.py), which uses [micro_ros_platformio/microros_utils/repositories.py](https://github.com/micro-ROS/micro_ros_platformio/blob/main/microros_utils/repositories.py).  The `.../repositories.py` was updated 3 weeks ago and mentions [deprecating foxy](https://github.com/micro-ROS/micro_ros_platformio/commit/88c4ea8e0b6448932c0847a9ef065c182b7c5439). 

The change I'm asking you to make to `firmware/platformio.ini` should checkout a release of that repo from last year, which included foxy support.

This would also explain why it worked for me with humble and rolling, as those aren't deprecated by micro_ros_platformio

justinbar, 2023-07-03:
it didn't work i got a lot of warning 
[image]

atticusrussell, 2023-07-03:
I can reproduce your errors (galactic branch, ROS foxy WSL Ubuntu 20.04):
```
atticus@AJR-L5Pro:~/github/linorobot2_hardware/firmware  (galactic)
$ pio run -e teensy40
Processing teensy40 (board: teensy40; platform: teensy; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/micro-ROS/micro_ros_platformio
git version 2.25.1
Cloning into '/home/atticus/.platformio/.cache/tmp/pkg-installing-9dd0chya'...
remote: Enumerating objects: 67, done.
remote: Counting objects: 100% (67/67), done.
remote: Compressing objects: 100% (51/51), done.
remote: Total 67 (delta 10), reused 31 (delta 5), pack-reused 0
Unpacking objects: 100% (67/67), 124.89 KiB | 1.07 MiB/s, done.
Library Manager: micro_ros_platformio@0.0.1+sha.35db5a9 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.18.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy @ 1.158.0 (1.58)
 - tool-teensy @ 1.158.0 (1.58)
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
Converting firmware.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing catkin-pkg with pip at PlatformIO environment
/home/atticus/.platformio/penv/bin/python -m pip install catkin-pkg
Collecting catkin-pkg
  Using cached catkin_pkg-0.5.2-py3-none-any.whl (76 kB)
Collecting docutils (from catkin-pkg)
  Downloading docutils-0.20.1-py3-none-any.whl (572 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 572.7/572.7 kB 5.2 MB/s eta 0:00:00
Collecting python-dateutil (from catkin-pkg)
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting pyparsing (from catkin-pkg)
  Downloading pyparsing-3.1.0-py3-none-any.whl (102 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 102.6/102.6 kB 12.7 MB/s eta 0:00:00
Requirement already satisfied: setuptools in /home/atticus/.platformio/penv/lib/python3.8/site-packages (from catkin-pkg) (44.0.0)
Collecting six>=1.5 (from python-dateutil->catkin-pkg)
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Installing collected packages: six, pyparsing, docutils, python-dateutil, catkin-pkg
Successfully installed catkin-pkg-0.5.2 docutils-0.20.1 pyparsing-3.1.0 python-dateutil-2.8.2 six-1.16.0
Installing lark-parser with pip at PlatformIO environment
/home/atticus/.platformio/penv/bin/python -m pip install lark-parser
Collecting lark-parser
  Downloading lark_parser-0.12.0-py2.py3-none-any.whl (103 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 103.5/103.5 kB 1.9 MB/s eta 0:00:00
Installing collected packages: lark-parser
Successfully installed lark-parser-0.12.0
Installing empy with pip at PlatformIO environment
/home/atticus/.platformio/penv/bin/python -m pip install empy
Collecting empy
  Using cached empy-3.3.4-py3-none-any.whl
Installing collected packages: empy
Successfully installed empy-3.3.4
Installing colcon-common-extensions with pip at PlatformIO environment
/home/atticus/.platformio/penv/bin/python -m pip install colcon-common-extensions
Collecting colcon-common-extensions
  Using cached colcon_common_extensions-0.3.0-py3-none-any.whl (6.5 kB)
Collecting colcon-cmake (from colcon-common-extensions)
  Using cached colcon_cmake-0.2.27-py3-none-any.whl (25 kB)
Collecting colcon-core (from colcon-common-extensions)
  Using cached colcon_core-0.12.1-py3-none-any.whl (126 kB)
Collecting colcon-defaults (from colcon-common-extensions)
  Using cached colcon_defaults-0.2.8-py3-none-any.whl (10.0 kB)
Collecting colcon-devtools (from colcon-common-extensions)
  Using cached colcon_devtools-0.2.3-py3-none-any.whl (7.5 kB)
Collecting colcon-library-path (from colcon-common-extensions)
[... 318 lines cut ...]
Compiling .pio/build/teensy40/liba2a/kinematics/kinematics.cpp.o
lib/imu/MPU9250.cpp: In member function 'bool MPU9250::writeMemoryBlock(const uint8_t*, uint16_t, uint8_t, uint8_t, bool, bool)':
lib/imu/MPU9250.cpp:2999:14: warning: 'progBuffer' may be used uninitialized in this function [-Wmaybe-uninitialized]
 2999 |     uint8_t *progBuffer;
      |              ^~~~~~~~~~
lib/imu/MPU9250.cpp:3025:23: warning: 'verifyBuffer' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3025 |         if (verify && verifyBuffer) {
      |                       ^~~~~~~~~~~~
lib/imu/MPU9250.cpp: In member function 'bool MPU9250::writeDMPConfigurationSet(const uint8_t*, uint16_t, bool)':
lib/imu/MPU9250.cpp:3104:81: warning: 'progBuffer' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3104 |                 if (sizeof(progBuffer) < length) progBuffer = (uint8_t *)realloc(progBuffer, length);
      |                                                                          ~~~~~~~^~~~~~~~~~~~~~~~~~~~
lib/imu/MPU9150.cpp: In member function 'bool MPU9150::writeMemoryBlock(const uint8_t*, uint16_t, uint8_t, uint8_t, bool, bool)':
lib/imu/MPU9150.cpp:3020:23: warning: 'verifyBuffer' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3020 |         if (verify && verifyBuffer) {
      |                       ^~~~~~~~~~~~
============================================== [FAILED] Took 205.42 seconds ==============================================

Environment    Status    Duration
-------------  --------  ------------
teensy40       FAILED    00:03:25.425
========================================== 1 failed, 0 succeeded in 00:03:25.425 ==========================================
```

I'll try to troubleshoot further

atticusrussell, 2023-07-03:
I got it to compile after some more troubleshooting. This repo needs some work to support different ROS distros and test them - I'll try to get that working if I can get PRs merged. In the meantime, @justinbar here's the workaround:

I found this existing issue https://github.com/linorobot/linorobot2_hardware/issues/13
and followed it.

my `firmware/platformio.ini` looks like this (in the relevant section):
```
[env]
platform = teensy
framework = arduino
upload_port = /dev/ttyACM0
upload_protocol = teensy-cli
board_microros_transport = serial
board_microros_distro = ${sysenv.ROS_DISTRO}
lib_deps =
  https://github.com/micro-ROS/micro_ros_platformio#1.0.0
  https://github.com/micro-ROS/micro_ros_utilities
build_flags = -I ../config
```
I didn't change anything above or below - just changed lib_deps
I then did the following, which you may even be able to just paste and have it work.

```
cd ~
git clone -b $ROS_DISTRO https://github.com/ros2/rosidl.git
cp -r rosidl/rosidl_typesupport_introspection_c linorobot2_hardware/firmware/.pio/libdeps/teensy40/
cd linorobot2_hardware/firmware/
pio run -e teensy40
```

this worked for me:
```
atticus@AJR-L5Pro:~/github/linorobot2_hardware/firmware  (galactic)
$ pio run -e teensy40
Processing teensy40 (board: teensy40; platform: teensy; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.18.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy @ 1.158.0 (1.58)
 - tool-teensy @ 1.158.0 (1.58)
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
Converting firmware.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/home/atticus/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/atticus/.platformio/penv/lib/python3.8/site-packages (6.0)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/atticus/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/atticus/.platformio/penv/lib/python3.8/site-packages (2.0.1)
Configuring teensy40 with transport serial
micro-ROS already built
Found 101 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.2aef8a2
|-- micro_ros_utilities @ 0.0.0+20230703120900.sha.5f1cead
|-- encoder
|-- imu
|-- kinematics
|-- motor
|-- odometry
|-- pid
Building in release mode
Compiling .pio/build/teensy40/src/firmware.ino.cpp.o
Compiling .pio/build/teensy40/lib3c7/micro_ros_utilities/type_utilities.c.o
[... 19 lines cut ...]
      |     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino:224:26: warning: ignoring return value of 'rcl_ret_t rcl_subscription_fini(rcl_subscription_t*, rcl_node_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  224 |     rcl_subscription_fini(&twist_subscriber, &node);
      |     ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino:225:18: warning: ignoring return value of 'rcl_ret_t rcl_node_fini(rcl_node_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  225 |     rcl_node_fini(&node);
      |     ~~~~~~~~~~~~~^~~~~~~
/home/atticus/github/linorobot2_hardware/firmware/src/firmware.ino:226:19: warning: ignoring return value of 'rcl_ret_t rcl_timer_fini(rcl_timer_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  226 |     rcl_timer_fini(&control_timer);
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
Linking .pio/build/teensy40/firmware.elf
Checking size .pio/build/teensy40/firmware.elf
Calculating size .pio/build/teensy40/firmware.elf
Building .pio/build/teensy40/firmware.hex
teensy_size: Memory Usage on Teensy 4.0:
teensy_size:   FLASH: code:116140, data:10428, headers:8596   free for files:1896452
teensy_size:    RAM1: variables:34368, code:114024, padding:17048   free for local variables:358848
teensy_size:    RAM2: variables:12416  free for malloc/new:511872
=============================================== [SUCCESS] Took 3.35 seconds ===============================================

Environment    Status    Duration
-------------  --------  ------------
teensy40       SUCCESS   00:00:03.349
=============================================== 1 succeeded in 00:00:03.349 ===============================================
```

justinbar, 2023-07-04:
think you it worked! so far I was able to drive my robot around. This project need to be up to date due these type of errors.

atticusrussell, 2023-07-04:
Glad to have helped! And agreed - there are a number of issues to be resolved. I only found this repo recently myself, but I made PR #34 to begin automatic testing for stuff like this, and if it gets merged by this repo's owner I plan to move on to fixes for issues like this.

atticusrussell, 2023-08-26:
@justinbar can you try checking out the [foxy](https://github.com/linorobot/linorobot2_hardware/tree/foxy) branch and confirm if it works now? Shouldn't need to tinker with anything. 

If it does work, do you mind closing this issue? Thanks !

justinbar, 2023-09-03:
it worked for foxy branch

atticusrussell, 2023-09-03:
@justinbar Wonderful. Can you close the issue?