linorobot/linorobot2_hardware #37: Issue while uploading firmware into teensy

phaneesh2, 2023-07-21:
I am using linorobot2 ROS2(Galactic) Ubuntu 20.04 Raspberrypi4 teensy3.1
[image]

pio run --target upload -e teensy31

While using above command i am facing issue as mentioned below

Downloading micro-ROS dev dependencies
KeyError: 'galactic':
  File "/home/ubuntu/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 175:
    env.SConscript("$BUILD_SCRIPT")

Help would be much appreciated.

phaneesh2, 2023-07-21:
> can you try the latest changes in

I tried with the above link.But still I am facing the same problem.

Looking for support from the team.

atticusrussell, 2023-07-21:
@grassjelly I troubleshot a similar issue in #35 . Would love to get your thoughts on PR #34, which could detect issues like these before a user runs into them and let maintainers know to address them. 

I'd love to work on a PR that fixes this issue and #35, but would want to know it would work and has a shot at being merged.

grassjelly, 2023-07-21:
hey @atticusrussell  commented on your PR

grassjelly, 2023-07-21:
@phaneesh2 could you post what's the error you're getting now?

atticusrussell, 2023-07-21:
> hey @atticusrussell  commented on your PR

@grassjelly I really don't wanna derail this issue off topic, but I can't see any comments from you on PR #34

If comment visibility continues to be an issue, maybe we discuss in a new issue in the repo to keep this one on track.

atticusrussell, 2023-07-21:
@phaneesh2 this is a guess and a temporary fix ( read (https://github.com/linorobot/linorobot2_hardware/issues/35#issuecomment-1617063865) and https://github.com/linorobot/linorobot2_hardware/issues/35#issuecomment-1618649572 for reason), but try the following for me:

In `firmware/platformio.ini`, try changing
`lib_deps = https://github.com/micro-ROS/micro_ros_platformio`
to 
`lib_deps = https://github.com/micro-ROS/micro_ros_platformio#1.0.0`

then do the following to install a dependency manually and try again:
```
cd ~
git clone -b $ROS_DISTRO https://github.com/ros2/rosidl.git
cp -r rosidl/rosidl_typesupport_introspection_c linorobot2_ws/linorobot2_hardware/firmware/.pio/libdeps/teensy31/
cd linorobot2_ws/linorobot2_hardware/firmware/
pio run -e teensy31
```

phaneesh2, 2023-07-22:
Thanks a lot. I reinstalled everything and it worked with the above link.
Thankyou for your support.

phaneesh2, 2023-07-22:
Hello All,

I had to come back again related to my earlier issue. Now everything is
configured properly with teensy. But some issue with micro ros - agent.

 ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1690032787.162839] error    | TermiosAgentLinux.cpp | init
    | open device error      | device: /dev/ttyACM0, errno: 16
Error while starting serial agent!

I tried to control it with a keyboard but the robot is not driving.

Looking forward to your reply.

Regards,
Haneesh P.

atticusrussell, 2023-08-26:
@phaneesh2 
Haneesh,
Can you test if the firmware will upload with the 'galactic' branch of this repo? Was recently updated and should work now.  If so, could you close this issue? 

Your second question should go in a new issue, and we can work it out from there .

Thanks

bibinthomas007, 2024-04-19:
> Hello All, I had to come back again related to my earlier issue. Now everything is configured properly with teensy. But some issue with micro ros - agent. ros2 run micro_ros_agent micro_ros_agent [...]

have you been able to sort it out.