linorobot/linorobot2_hardware #56: Iron branch

atticusrussell, 2023-08-26:
The newest ROS2 distro is Iron. I think a branch should be added for Iron, and it should be added to CI (pretty easy just following the end of the README). The only thing the README leaves out is modifying the README, which is as simple as adding another row to the table and changing the words and urls to reference the new branch.

grassjelly, 2023-08-27:
shall we just use rename rolling to iron? I think having two different branches that does the same thing could confuse users

atticusrussell, 2023-08-27:
I think rolling should always exist and have features developed on it because it will always be up to date with the bleeding edge of ROS2 . 

I envision users of linorobot2_hardware simply checkout the branch that they are using for their project, (which will rarely be rolling which AFAIK exists mostly for development sake). Most projects right now are on humble, but I can imagine some being on Iron in the near future.

In reality, the iron branch will likely be exactly the same as Rolling and as Humble for a while, however my reasoning in suggesting it in addition to the existing branches is twofold:

1. I think it's existence actually simplifies what branch to use for users, as if they are on iron it's quite obvious to use the iron branch.

2. Its existence will mean CI is checking that the dependencies (platformio libdeps), one of which (micro ros utils I think) queries the ros distro and chooses it's branch accordingly, and we can test that everything works smoothly for users on iron.