linorobot/linorobot2_hardware #61: Platformio install command stopped working

Russ76, 2023-08-27:
The included command line for installing Platformio no longer works. Did they move things in their repo?

atticusrussell, 2023-08-27:
I think PlatformIO may have changed their install script's url. Give this one a try and let me know how it works for you:
```
python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py)"
```

https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html

If it works, would you want to submit a PR updating the command in the README?
If not, I would also be happy to.

atticusrussell, 2023-10-30:
@Russ76 you mind closing this issue if appropriate? Pretty sure #66 resolved this.

Russ76, 2023-11-01:
Yes, I went to their web page and followed directions there! Thanks

closed 2023-11-01 by Russ76