linorobot/linorobot2_hardware #64: Compiling errors trying to program Teensy4.0

Russ76, 2023-09-08:
This is on a new user set up to test issue. Ubuntu Jammy, Ros Humble, Robot computer Ryzen 7 with 16 GB RAM.
Would verbose output be helpful?

 pio run --target upload -e teensy40
Processing teensy40 (board: teensy40; platform: teensy; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.18.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.158.0 (1.58) 
 - tool-teensy @ 1.158.0 (1.58) 
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
Converting firmware.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/home/mowbot/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/mowbot/.platformio/penv/lib/python3.10/site-packages (6.0.1)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/mowbot/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/mowbot/.platformio/penv/lib/python3.10/site-packages (2.0.1)
Configuring teensy40 with transport serial
micro-ROS already built
Found 99 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.35db5a9
|-- encoder
|-- imu
|-- kinematics
|-- motor
|-- odometry
|-- pid
Building in release mode
Compiling .pio/build/teensy40/src/firmware.ino.cpp.o
Archiving .pio/build/teensy40/lib2dc/libWire.a
Indexing .pio/build/teensy40/lib2dc/libWire.a
Compiling .pio/build/teensy40/lib1ce/imu/MPU9250.cpp.o
Compiling .pio/build/teensy40/lib755/kinematics/kinematics.cpp.o
Compiling .pio/build/teensy40/libbea/Servo/Servo.cpp.o
Compiling .pio/build/teensy40/lib098/odometry/odometry.cpp.o
Compiling .pio/build/teensy40/lib5c3/pid/pid.cpp.o
Compiling .pio/build/teensy40/FrameworkArduino/AudioStream.cpp.o
Compiling .pio/build/teensy40/FrameworkArduino/CrashReport.cpp.o
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware (copy).ino:46:17: error: redefinition of 'rcl_publisher_t odom_publisher'
   46 | rcl_publisher_t odom_publisher;
      |                 ^~~~~~~~~~~~~~
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware.ino:46:17: note: 'rcl_publisher_t odom_publisher' previously declared here
   46 | rcl_publisher_t odom_publisher;
      |                 ^~~~~~~~~~~~~~
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware (copy).ino:47:17: error: redefinition of 'rcl_publisher_t imu_publisher'
   47 | rcl_publisher_t imu_publisher;
      |                 ^~~~~~~~~~~~~
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware.ino:47:17: note: 'rcl_publisher_t imu_publisher' previously declared here
   47 | rcl_publisher_t imu_publisher;
      |                 ^~~~~~~~~~~~~
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware (copy).ino:48:20: error: redefinition of 'rcl_subscription_t twist_subscriber'
   48 | rcl_subscription_t twist_subscriber;
      |                    ^~~~~~~~~~~~~~~~
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware.ino:48:20: note: 'rcl_subscription_t twist_subscriber' previously declared here
   48 | rcl_subscription_t twist_subscriber;
      |                    ^~~~~~~~~~~~~~~~
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware (copy).ino:50:25: error: redefinition of 'nav_msgs__msg__Odometry odom_msg'
   50 | nav_msgs__msg__Odometry odom_msg;
      |                         ^~~~~~~~
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware.ino:50:25: note: 'nav_msgs__msg__Odometry odom_msg' previously declared here
   50 | nav_msgs__msg__Odometry odom_msg;
[... 267 lines cut ...]
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware (copy).ino:225:18: warning: ignoring return value of 'rcl_ret_t rcl_node_fini(rcl_node_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  225 |     rcl_node_fini(&node);
      |     ~~~~~~~~~~~~~^~~~~~~
/home/mowbot/lino2_ws/src/linorobot2_hardware/firmware/src/firmware (copy).ino:226:19: warning: ignoring return value of 'rcl_ret_t rcl_timer_fini(rcl_timer_t*)' declared with attribute 'warn_unused_result' [-Wunused-result]
  226 |     rcl_timer_fini(&control_timer);
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
*** [.pio/build/teensy40/src/firmware.ino.cpp.o] Error 1
lib/imu/MPU9250.cpp: In member function 'bool MPU9250::writeMemoryBlock(const uint8_t*, uint16_t, uint8_t, uint8_t, bool, bool)':
lib/imu/MPU9250.cpp:2999:14: warning: 'progBuffer' may be used uninitialized in this function [-Wmaybe-uninitialized]
 2999 |     uint8_t *progBuffer;
      |              ^~~~~~~~~~
lib/imu/MPU9250.cpp:3025:23: warning: 'verifyBuffer' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3025 |         if (verify && verifyBuffer) {
      |                       ^~~~~~~~~~~~
lib/imu/MPU9250.cpp: In member function 'bool MPU9250::writeDMPConfigurationSet(const uint8_t*, uint16_t, bool)':
lib/imu/MPU9250.cpp:3104:81: warning: 'progBuffer' may be used uninitialized in this function [-Wmaybe-uninitialized]
 3104 |                 if (sizeof(progBuffer) < length) progBuffer = (uint8_t *)realloc(progBuffer, length);
      |                                                                          ~~~~~~~^~~~~~~~~~~~~~~~~~~~
========================== [FAILED] Took 4.65 seconds ==========================

Environment    Status    Duration
-------------  --------  ------------
teensy40       FAILED    00:00:04.647
==================== 1 failed, 0 succeeded in 00:00:04.647 ====================
~/lino2_ws/src/linorobot2_hardware/firmware$

grassjelly, 2023-09-08:
Could you try a fresh setup anywhere outside your ROS2 workspace?

Russ76, 2023-09-09:
Yes, now it's working! Thanks

grassjelly, 2023-09-10:
Great! Closing this

closed 2023-09-10 by grassjelly