linorobot/linorobot2_hardware #88: MicroRos Agent Stucked

AdibSuid, 2024-05-19:
I succesfully uploaded the firmware into teensy 4.0, but then when I run micro-ros-agent, it won't start any node as attached picture below:[image]

I have tried to push the reset button then it become like this:[image]

I also tried to unplug and plugged again teensy board then it become like this:[image]

I'm using ros2 humble, but I already tried on ros2 galactic also. Already tried both on laptop and Raspberry Pi 4B.

hippo5329, 2024-05-24:
Please check your USB connection.
1. lsusb
2. ls /dev/tty*

hippo5329, 2024-05-24:
If you are new to micro-ROS, you may use pico and esp32 instead of teensy.  Please check my wiki at https://github.com/hippo5329/linorobot2_hardware/wiki .

AdibSuid, 2024-05-25:
> If you are new to micro-ROS, you may use pico and esp32 instead of teensy.  Please check my wiki at https://github.com/hippo5329/linorobot2_hardware/wiki .

can I use teensy and for your github repo?

hippo5329, 2024-05-25:
Yes, it should work. I suggested esp32 and pico because they are cheaper than teensy and have hardware encoder counters instead of the interrupt-driven on teensy. I have tested upload and connect to micro-ROS agent with teensy40 and teensy41. Though I do not use teensy to build robots and have not test teensy on real robots. There are issues in platformio teensy upstream, so the first upload to teensy might fail. But the second upload should success. It is not related to linorobot. You may use lino_base_config.h as a template and create your custom configuration. I have suggested test the micro-ROS agent connection with the micro-controller module alone before wiring and building the robot. BTW, Can you see the /dev/ttyACM0 device after upload?

hippo5329, 2024-05-25:
## Test micro-ROS agent connection - Recommended

It is recommended to test the micro-ROS agent connection with the micro-controller module alone before wiring and building the robots. Choose the configuration you will use to build your robot. If you use teensy and lino_base_config, comment out the "#define USE_GY85_IMU" in lino_base_config.h.

Then test the micro-ROS agent connection.

hippo5329, 2024-05-25:
Esp32 and pico are suggested because they are cheaper than teensy and have hardware encoder counters instead of the interrupt-driven on teensy. The hardware encoder counters can count up to MHz. It will be almost impossible for the more expensive teensy to reach this speed.

AdibSuid, 2024-05-26:
> ## Test micro-ROS agent connection - Recommended
> 
> It is recommended to test the micro-ROS agent connection with the micro-controller module alone before wiring and building the robots. Choose the configuration you will use to build your robot. If you use teensy and lino_base_config, comment out the "#define USE_GY85_IMU" in lino_base_config.h.
> 
> Then build, upload and test the micro-ROS agent connection.

I see, alright thanks

AdibSuid, 2024-05-26:
> ## Test micro-ROS agent connection - Recommended
>
> It is recommended to test the micro-ROS agent connection with the micro-controller module alone before wiring and building the robots. Choose the [...]

May I know why need to comment out the IMU define?

hippo5329, 2024-05-26:
We would like the teensy module connects to micro-ROS agent without the IMU. If the IMU is enabled but failed to initialize, the teensy will stop and blink the LED 3 times per second. So we comment out the IMU. Only the teensy module alone is needed to perform this connection test.

graham-fleming, 2024-05-26:
I am also having these problems.

Motor / encoder checks work - teensy40 firmware upload works

Is running the micro_ros_agent supposed to have more output? 

ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0

Commenting out the IMU shows error about commenting out IMU

I am using IMU GY-521

[image]

[image]

hippo5329, 2024-05-26:
Please switch to my fork.

cd linorobot2_hardware
git remote add hippo https://github.com/hippo5329/linorobot2_hardware.git
git fetch hippo
git reset --hard
git checkout -b hippo hippo/master

Comment out the GY85 in line_base_config.h, build, upload and test connection.

hippo5329, 2024-05-26:
Gy-521 is MPU6050.

graham-fleming, 2024-05-26:
Oh nice I switched over to your fork and it seems to be working now 

There is more output after running micro_ros_agent

Teensy has solid light - keys pressed for teleop_twist move wheels

Will investigate further

Thank you @hippo5329 for the work

[image]