linorobot/linorobot2_hardware #93: WAITING FOR ROBOT_DESCRIPTION TO BE PUBLISH...

thanhtruongphan, 2024-11-01:
[image]

follow this [instructions, ](https://github.com/linorobot/linorobot2?tab=readme-ov-file#11a-using-a-real-robot)

I have a 2wd + A1M8 + IMU (not camera)
everything is ok, but until I run this command line **```ros2 launch linorobot2_bringup bringup.launch.py```**, the result is like in the picture... 

_**[joint_state_publisher-3] [INFO] [1730453364.657480572] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...**_ 

can you help me @grassjelly 
Thank you so much 👍

thanhtruongphan, 2024-11-01:
and cannot control manual using **```teleop_twist_keyboard```** 
[image]

thanhtruongphan, 2024-11-01:
and when I run **```ros2 launch linorobot2_gazebo gazebo.launch.py```**:

[image]

atticusrussell, 2025-04-12:
ran into the same issue today - the issue isn't with linorobot - it's a bad log level description in joint state publisher
https://github.com/ros/joint_state_publisher/issues/113

PaulBouchier, 2025-12-25:
Closing per @atticusrussell recommendation